frickelzeugs/FingerprintDoorbell #3: Offtopic: NS-Panel

jhartlep, 2022-10-23:
Moin,
zunächst mal eine Hammer Idee mit der Klingel! Bei Spiel-und-Zeug hattest Du in dem Video aber auch noch Dein NS-Panel vorgestellt.
Hast Du hierfür auch eine Dokumentation? Das täte mich brennend für mein Panel interessieren.

Danke und Gruß

mendikant, 2022-10-24:
Moin, und ich fände ne Doku zu der Bettwaage spannend... 

Und: Wirklich coole Ideen... 

Danke und Gruß

Beast-Panik, 2022-10-24:
Um das NSPanel konfigurierbar zu machen gibt es hier den ersten einstieg 
https://blakadder.com/nspanel-hacking/
Für eine Einbindung unter IOBroker z.B. gibt es dann auch hier noch einen sehr Aktiven Topic für den Einstieg.
https://forum.iobroker.net/topic/58170/sonoff-nspanel-mit-lovelace-ui
So läuft es bei mir z.B.

jhartlep, 2022-10-24:
Grundsätzlich hab ich das NS-Panel bei mir schon drin im iobroker. Das ist nicht das Problem. Nur hat mir hier die UI, die man im Video gesehen hat, einfach fantastisch gefallen.

frickelzeugs, 2022-10-26:
Das Timing der Veröffentlichung des Spielundzeugs Videos war etwas ungünstig, sind direkt danach in Urlaub geflogen ;-) 
Die Frage wurde hier auch schon gestellt: http://www.frickelzeugs.de/forums/topic/sonoff-ns-panel/

Wenn ich nächste Woche wieder zurück bin, kann ich gerne die TFT (UI-Files des Nextion Displays) hier bei Github hochladen. Nur mit einer detaillierten Anleitung will ich nicht allzuviel Hoffnung machen. Das erstellen von Custom UIs für das NSPanel ist echt frickelig.

Bei der Bettwaage fällt die Anleitung deutlich kürzer aus, das ist eigentlich nur Standard-Tasmota auf einem ESP8266 mit entsprechender Konfiguration. Da kann ich auf meinem Blog kurz noch was zusammenschreiben...

jhartlep, 2022-10-26:
Hey, super! Danke für die Rückmeldung.
Bin mal auf Deine Lösung gespannt im Vergleich zu meiner bisherigen LoveLaceUI-Lösung.

frickelzeugs, 2022-11-03:
@jhartlep Schau mal, ob du was damit anfangen kannst: https://github.com/frickelzeugs/nspanel-frickelui

closed 2023-01-13 by frickelzeugs