frickelzeugs/FingerprintDoorbell #7: Reboot Loop after WIFI credentials are saved

janiskelemen, 2022-10-25:
Hi,

first of all thanks for making your FingerprintDoorbell OS! I have a similar Project but with a dedicated Ringer Button and was eager to try out your approach (especially since you put in the work and creating an additional Fingerprint DB :D ). However, I tried to install the precompiled files which worked fine but when I try to compile the files using PlatformIO my ESP gets stuck in a reboot loop after I entered my WIFI credentials and saved them. One thing I changed to get the initial version running was to use the dev version of ESPAsyncWebserver in order to get it compiled. Did you ever have this reboot loop issue?

`[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
monitor_speed = 115200
lib_deps = 
	**https://github.com/me-no-dev/ESPAsyncWebServer.git**
	ayushsharma82/AsyncElegantOTA@^2.2.6
	knolleary/PubSubClient@^2.8
	adafruit/Adafruit Fingerprint Sensor Library@^2.0.7
	intrbiz/Crypto@^1.0.0
lib_ldf_mode = deep+`

frickelzeugs, 2022-10-26:
No, not that I can remember. But I haven't tried to recompile this project since somewhen in May(?). Maybe things or dependencies have changed. I'm on vacation this week and will try to recompile when back.

janiskelemen, 2022-11-03:
Hi @frickelzeugs did you have a chance to try to compile it? :)

janiskelemen, 2022-11-07:
Just tried your recent commit, and it solved the reboot issue. Thanks a million @frickelzeugs! :)

closed 2022-11-07 by janiskelemen