frickelzeugs/FingerprintDoorbell #10: Compiling fail - xtensa/hal.h

Fillip38, 2022-10-26:
Good evening, 
I just tried to compile this via VSCode, but I get the following error message there
Can anyone help me with this?

`Processing esp32doit-devkit-v1 (platform: espressif32; board: esp32doit-devkit-v1; framework: arduino)
----------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif32/esp32doit-devkit-v1.html
PLATFORM: Espressif 32 (5.2.0) > DOIT ESP32 DEVKIT V1
HARDWARE: ESP32 240MHz, 320KB RAM, 4MB Flash
DEBUG: Current (cmsis-dap) External (cmsis-dap, esp-bridge, esp-prog, iot-bus-jtag, jlink, minimodule, olimex-arm-usb-ocd, olimex-arm-usb-ocd-h, olimex-arm-usb-tiny-h, olimex-jtag-tiny, tumpa)
PACKAGES:
 - framework-arduinoespressif32 @ 3.20005.220925 (2.0.5)
 - tool-esptoolpy @ 1.40201.0 (4.2.1)
 - toolchain-xtensa-esp32 @ 8.4.0+2021r2-patch3
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ deep, Compatibility ~ soft
Found 38 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP Async WebServer @ 1.2.3
|   |-- AsyncTCP @ 1.1.1
|   |-- AsyncElegantOTA @ 2.2.7
|   |-- FS @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- AsyncElegantOTA @ 2.2.7
|-- PubSubClient @ 2.8.0
|-- Adafruit Fingerprint Sensor Library @ 2.1.0
|-- Crypto @ 1.0.0
|-- DNSServer @ 2.0.0
|   |-- WiFi @ 2.0.0
|-- SPIFFS @ 2.0.0
|   |-- FS @ 2.0.0
|-- WiFi @ 2.0.0
|-- Preferences @ 2.0.0
Building in release mode
Compiling .pio\build\esp32doit-devkit-v1\src\FingerprintManager.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\src\SettingsManager.cpp.o
Compiling .pio\build\esp32doit-devkit-v1\src\main.cpp.o
In file included from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig_arch.h:103,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/esp_additions/freertos/FreeRTOSConfig.h:79,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:57,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from .pio/libdeps/esp32doit-devkit-v1/Adafruit Fingerprint Sensor Library/Adafruit_Fingerprint.h:8,
                 from src/FingerprintManager.h:4,
                 from src/FingerprintManager.cpp:1:
C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:10: fatal error: xtensa/hal.h: No such file or directory     
 #include <xtensa/hal.h>
          ^~~~~~~~~~~~~~
compilation terminated.
In file included from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig_arch.h:103,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/esp_additions/freertos/FreeRTOSConfig.h:79,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:57,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/libraries/Preferences/src/Preferences.h:17,
                 from src/SettingsManager.h:4,
                 from src/SettingsManager.cpp:1:
C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:10: fatal error: xtensa/hal.h: No such file or directory     
 #include <xtensa/hal.h>
          ^~~~~~~~~~~~~~
compilation terminated.
Compiling .pio\build\esp32doit-devkit-v1\lib589\AsyncTCP\AsyncTCP.cpp.o
*** [.pio\build\esp32doit-devkit-v1\src\SettingsManager.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\src\FingerprintManager.cpp.o] Error 1
In file included from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig_arch.h:103,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/esp_additions/freertos/FreeRTOSConfig.h:79,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:57,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/esp_event/include/esp_event.h:20,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:27,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:28,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:32,
                 from src/main.cpp:5:
C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:10: fatal error: xtensa/hal.h: No such file or directory     
 #include <xtensa/hal.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\src\main.cpp.o] Error 1
In file included from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/FreeRTOSConfig_arch.h:103,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/esp_additions/freertos/FreeRTOSConfig.h:79,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/include/freertos/FreeRTOS.h:57,
                 from C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/cores/esp32/Arduino.h:33,
                 from .pio/libdeps/esp32doit-devkit-v1/AsyncTCP/src/AsyncTCP.cpp:22:
C:/Users/LauLip/.platformio/packages/framework-arduinoespressif32/tools/sdk/esp32/include/freertos/port/xtensa/include/freertos/xtensa_config.h:40:10: fatal error: xtensa/hal.h: No such file or directory     
 #include <xtensa/hal.h>
          ^~~~~~~~~~~~~~
compilation terminated.
*** [.pio\build\esp32doit-devkit-v1\lib589\AsyncTCP\AsyncTCP.cpp.o] Error 1
========================================================================================== [FAILED] Took 6.27 seconds ==========================================================================================`

henfri, 2022-10-27:
Hello,

this may not help you directly, but I also compiled with platformio/VSCode. I also had numerous issues, but not this one...
So, it may be an issue with your particular configuration of platformio and not with the code.

I suggest you g google this error.

Best regards,
Hendrik

Fillip38, 2022-10-27:
hello, 
that's correct, I installed the whole thing again freshly on another pc, so the compilation doesn't run with the error, but with the one for OTA, which was already addressed in the other topic

closed 2022-10-27 by Fillip38