frickelzeugs/FingerprintDoorbell #9: Not compiling | fatal error: Update.h: No such file or directory

kimstade, 2022-10-26:
Hi,

unfortunately the project isn't compiling in visual studio (with no editing in code, 100% original git clone)
-> https://imgur.com/a/SoOw1IF

I've tried different versions of the libraries and different versions of the platform io extension.
But everything leads to the same problem :(

Yes, I could flash the precompiled one, and yes it's working. But for my needs I have to change the project a bit.
Would be really nice if it would compile.

frickelzeugs, 2022-10-26:
Hi kimstade, seems like there is a problem with the OTA updater library I am using, will check this next week after my vacation.

kimstade, 2022-10-27:
Hi henfri,

I've tried to load the AsyncElegantOTA lib via github before without any changes. But I can confirm that loading both the ESP Async WebServer AND the AsyncElegantOTA libs via github is resolving this issue. Compiling and flashing now works without any problems. Thank you!

closed 2022-10-27 by kimstade

henfri, 2022-10-27:
:-)

kimstade, 2022-10-27:
Unfortunately the esp32 won't connect to my wifi anymore. After flashing, the access point comes up, I can put in the settings and the ap closes. But then it won't connect. After rebooting the fingerprint led stays dark. Reflashing leads to the same behavior (without any changes to the project except the platformio.ini) :(

After that I've tested the precompiled firmware again. That's working perfectly fine.

kimstade, 2022-10-27:
Update: With your fork it's working fine :)

henfri, 2022-10-27:
Hello,

I am a bit lost: the esp32 won't connect with your self-compiled firmware that you created by using the original code and fixed the compilation errors by applying my suggestion from
https://github.com/frickelzeugs/FingerprintDoorbell/issues/9#issuecomment-1293099511

Right?

Then, the precompiled FW and my fork are working fine?

That is odd. I have only changed one more line that is wifi related, I think.

kimstade, 2022-10-27:
Yes, you are right

My self-compiled fw with your suggested fix: compiling works, wifi not connecting
Precompiled fw from frickelzeugs: wifi works

Your fork self-compiled: compiling works, wifi works, everythings fine

eingabe, 2022-11-04:
> Hallo, ich habe diesen Fehler behoben: in platformio.ini:
> 
> ```
> lib_deps = 
> 	me-no-dev/ESP Async WebServer@^1.2.3 # this should fix it https://github.com/khoih-prog/ESPAsyncWebServer/archive/refs/heads/master.zip.
> 	https://github.com/ayushsharma82/AsyncElegantOTA/archive/refs/heads/master.zip # ayushsharma82/AsyncElegantOTA@^2.2.6 fails with update.h missing
> 	knolleary/PubSubClient@^2.8
> 	adafruit/Adafruit Fingerprint Sensor Library@^2.0.7
> 	intrbiz/Crypto@^1.0.0
> ```

Hello, after that option i can compile the code. But Flash over platform io or OTA is not running.
I must erase the flash and start at the beginning.
also i can not flash without any changing from the code only the original bin files.
but it was a perfect project, i am so happy if it's work for me

henfri, 2022-11-05:
> But Flash over platform io or OTA is not running.

That is a very unspecific problem description.

> I must erase the flash and start at the beginning.

And what happens if you don't?

>  i can not flash without any changing from the code only the original bin files

Again: Too unspecific.

eingabe, 2022-11-07:
Hallo, wie ich gesehen habe hast du ein paar Dinge verändert. Und ich kann dir sagen, dass jetzt alles funktioniert. Danke ich teste alles weitere :-)