frickelzeugs/FingerprintDoorbell #14: 404 on Wifi Setup

henfri, 2022-10-30:
Hello,

unfortunately, I get a 404 Error when connecting to 192.168.4.1 (of course I have connected to the right WiFi before).

```
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6360
entry 0x400806b4
[E][Preferences.cpp:49] begin(): nvs_open failed: NOT_FOUND

Adafruit finger detect test
Found fingerprint sensor!
Reading sensor parameters
Status: 0x4
Sys ID: 0x0
Capacity: 200
Security level: 3
Device address: FFFFFFFF
Packet len: 128
Baud rate: 57600
Sensor contains 0 templates
loadFingerListFromPrefs
0 fingers loaded from preferences.
Started WiFi-Config mode
AP IP address: 192.168.4.1
Failed to obtain time
[no time]: System booted successfully!
```

The only odd thing in the log is the` [E][Preferences.cpp:49] begin(): nvs_open failed: NOT_FOUND`.
But I do not think, this is related, is it?

Greetings,
Hendrik

frickelzeugs, 2022-10-30:
Are you still trying to use a ESP8266? Preferences (storing data in NVS) is a feature of ESP32 and not available on the ESP8266 and thereby the complete storage layer has to be ported to something available on the ESP8266.

henfri, 2022-10-30:
No, I am using ESP32.
@all: can you please check, if you also have the` [E][Preferences.cpp:49] begin(): nvs_open failed: NOT_FOUND.` to rule out that this is the issue?

henfri, 2022-10-30:
I have now checked with your original code, except for my changes in `platformio.ini`

```
[env:esp32doit-devkit-v1]
platform = espressif32@3.5.0      # @3.5.0 #otherwise Update.h is missing. See https://community.platformio.org/t/update-espressif-32-version-4-1-0-leads-to-error-collect2-exe-error-ld-returned-1-exit-status/27742/3 
                                  # Other fix is to use a different version of the Asyncwebserver (https://github.com/khoih-prog/ESPAsyncWebServer/archive/refs/heads/master.zip.)
board = nodemcu-32s
board_build.flash_mode= qio  # fixes SPIFFS error
framework = arduino
monitor_speed = 115200
lib_deps = 
	me-no-dev/ESP Async WebServer@^1.2.3 # this should fix it https://github.com/khoih-prog/ESPAsyncWebServer/archive/refs/heads/master.zip.
	https://github.com/ayushsharma82/AsyncElegantOTA/archive/refs/heads/master.zip # ayushsharma82/AsyncElegantOTA@^2.2.6 fails with update.h missing
	knolleary/PubSubClient@^2.8
	adafruit/Adafruit Fingerprint Sensor Library@^2.0.7
	intrbiz/Crypto@^1.0.0
lib_ldf_mode = deep
```

I get the `[E][Preferences.cpp:49] begin(): nvs_open failed: NOT_FOUND` also here.

And also, I get the 404. So, it is not related to my changes.

henfri, 2022-10-31:
Hello,

I got it working by flashing your binary. That worked so far, that I got the wifi configuration page (I did not configure).
Then, I flashed my binary again and it worked as well. 

Maybe, there was something wrong with my partitioning/bootloader etc.

Greetings,
Hendrik

closed 2022-10-31 by henfri

henfri, 2022-10-31:
Hello,

any idea how to improve this? After erasing the flash, I always have to first to flash your binary before I can flash the pio code.
Is maybe the building and flashing of the SPIFFS image missing in the PIO chain?
I am struggling with the same issue with the ESP8266 port now.

Greetings,
Hendrik

henfri, 2022-10-31:
Ah... Maybe it is user error...
There is a task "build filesystem" and "upload filesystem". I suppose, I would have to run that...?

frickelzeugs, 2022-10-31:
Yes 😉