frickelzeugs/FingerprintDoorbell #15: esp8266 port

henfri, 2022-10-30:
Are you still trying to use a ESP8266? Preferences (storing data in NVS) is a feature of ESP32 and not available on the ESP8266 and thereby the complete storage layer has to be ported to something available on the ESP8266.

_Originally posted by @frickelzeugs in https://github.com/frickelzeugs/FingerprintDoorbell/issues/14#issuecomment-1296302104_
     

I am using https://github.com/vshymanskyy/Preferences for ESP8266, which makes it available also on this hardware.

PhagoZ, 2022-11-16:
Ich hab das eben mal auf nem 8266 Di mini geflashed
Das config Wifi wird zwar erstellt, aber es gibt keine weboberfläche.
Ich dachte erst es würde an der IP liegen (einer meiner Router hat die selbe IP), allerdings war das Problem noch da, nachdem ich die IP für die config geändert habe.

bgirr, 2022-11-16:
Hatte das gleiche Problem, liegt tatsächlich am ESP8266. Problem ist nicht so einfach lösbar, bin deswegen jetzt auf einen ESP32 umgestiegen.

PhagoZ, 2022-11-16:
ach mist, von den 8266ern hab ich sooo viele rumzuliegen ^^

henfri, 2022-11-16:
Don't give up so easily.
1) what code are you using?
2) did you upload the Filesystem image?
3) what do you see on the serial-log?

Best regards,
Hendrik

PhagoZ, 2022-11-16:
> Don't give up so easily.
> 
> 1. what code are you using?
I copied the code from:
[https://github.com/frickelzeugs/FingerprintDoorbell/pull/11/commits/0930022173655ad90a6e52927eae8cdc78cec440](url)
> 2. did you upload the Filesystem image?
I uloaded the code via Visual Studio.. so do I??
Didn´t find a way to do it.
> 3. what do you see on the serial-log?
The log from VS during the upload?
> 
> Best regards, Hendrik

PhagoZ, 2022-11-18:
Yeah I already found the upload Filesystem and did it.

I will try again on monday and Post the log.
For now I ordered some D1 Minis with esp32 and it's running.
But I will help with the Port to 8266 with my "Problems".

henfri, 2022-11-20:
Yes, would be nice to get this running also on 8266.

RocketSience, 2022-12-09:
@henfri 

1. You are using Serial1 to communicate with the fingerprint reader in your code.

[image]

As Serial1 on an ESP8266 can only transmit data, how can this work?
How did you connect the reader?

I am trying your fork, but get no RX connection from the fingerprint reader, however TX ist working (LED Colour changes)

2. i have the same problem as @PhagoZ. I have uploaded the Filesystem, but there are no Files listed...
[image]

henfri, 2023-05-03:
Not that i know of

AxelHoss, 2023-05-04:
Ok, then i will try it with tasmota