frickelzeugs/FingerprintDoorbell #23: Kinderfinger / Kleine Finger werden nicht erkannt

PinkyFooFoo, 2022-11-06:
Guten Abend zusammen,

auch von mir meinen größten Respekt für dieses richtig coole Projekt.
Setup (Hard/Software) funktionierte 1a. Bin gerade dabei das ganze zu verlegen. Dabei ist mir aufgefallen, dass der Fingerabdruck-Sensor die Finger der Kinder (7-9 Jahre) nicht erkennt. 

-> Start Enrollment:
Take #1 (place your finger on the sensor until led ring stops flashing, then remove it).
Take #2 (place your finger on the sensor until led ring stops flashing, then remove it).
Take #3 (place your finger on the sensor until led ring stops flashing, then remove it).
Take #4 (place your finger on the sensor until led ring stops flashing, then remove it).
Take #5 (place your finger on the sensor until led ring stops flashing, then remove it).
Enrollment failed. (Code 1)
 ScanResult Error (Code 6)

Dass ein Finger auf dem Sensor liegt erkennt der Sensor, aber er erkennt ihn nicht als "Fingerabdruck". Bei der großen funktioniert immerhin schonmal der Daumen. 

Da ich davon ausgehe, dass bei dir auch Kinder rumlaufen - hast du das Problem in den Griff bekommen? Oder sind deine Kinder schon älter, sodass die Finger "durchgehen"?

Gruß

frickelzeugs, 2022-11-06:
Hi,

meine Jungs sind 11 und 13 und haben daher schon etwas dickere Finger, aber der Anlernprozess klappte auch da nicht immer gleich auf Anhieb. Tipp: Versuch mal die Finger möglichst zügig und immer gleich auf den Sensor zu legen und nach dem Auflegen still zu halten bis gelesen wurde. Beim Handy ist man ja gewohnt den Finger immer in möglichst vielen unterschiedlichen Posen auf den Sensor zu legen, das ist bei dem Sensor hier aber leider kontraproduktiv, der mag es möglichst gleich. Code 6 entspricht laut der Adafruit Library = "Image mess", also unsauberes Bild, evtl. nicht still genug gehalten?
Ich könnte die Anzahl der Durchgänge beim Anlernen (aktuell fix auf 5 kodiert) auch auf bis zu 2 reduzieren (konfigurierbar über die UI). Das macht das Anlernen dann einfacher, aber die Erkennung hinterher wieder etwas weniger zuverlässig.

PinkyFooFoo, 2022-11-06:
Ah "Image Mess" hilft als Info. Wir versuchens mal mit "immer gleich drauflegen". Mal sehen ob uns das weiterbringt. 
Die Möglichkeit "vorübergehend" auf 2 Auflagen zu reduzieren (für beispielsweise Kinderfinger) wäre natürlich grundsätzlich cool, wenn man das über die UI einstellen kann.

PinkyFooFoo, 2022-11-09:
So, da bin ich wieder. Hat etwas gedauert - aber ich habe inzwischen alles ausprobiert. Die Kinderfinger werden einfach nicht erkannt. Ich erhalte immer wieder den Hinweis wie oben. 

Wirklich schade, denn wenn die Kinder den Sensor nicht nutzen können ist das ganze Projekt nicht so richtig sinnvoll. Für Erwachsene läufts mega gut!

Könntest du eine Option einbauen, dass man die Anzahl der Durchgänge konfigurieren kann? Dann würde ich es damit nochmal versuchen.

Interessant: Wenn ich meinen Finger anlerne und ihn einfach auf dem Sensor liegen lasse, dann erkennt der Sensor durchweg den Finger und "wartet" mit dem Anlernprozess bis ich den Finger wieder hochnehme und dann wieder ablege. 
Bei meinen Kindern habe ich den Versuch gestartet den Finger einfach drauf liegen zu lassen. (Damit der Finger wirklich ruhig genug liegt und es kein "Image mess" geben kann.) Der Finger wird vom Sensor dann zwischendrin als "hochgehoben" erkannt obwohl er weiter drauf liegt. Ich vermute, das Gesamtbild ist dem Sensor zu klein und er wertet es als "kein Fingerabdruck".

PhagoZ, 2022-11-19:
Ich hab das ganze gestern mit unserer 4 jährigen probiert.
Habe beide Daumen eingelernt bekommen und sie werden auch erkannt recht zuverlässig erkannt. 
Aufgefallen ist mir, daß es bei kleinen dingern besser klappt, wenn man den Finger etwas weiter oben ansetzt. 
Es wird also näher am Gelenk bescannt. Das funktioniert (im Rahmen der aufmerksamkeitsspanne einer 4jährigen) ganz gut.

PinkyFooFoo, 2023-01-14:
Spannend. Ich habe zwischenzeitlich einen weiteren Fingerabdruckscanner bestellt, habe insgesamt also 3 Stück hier ausprobiert. Alle drei erkennen von meiner 9 Jährigen: Daumen und Zeigefinger, von meiner 7 Jährigen: Garkeinen Finger. Aber auch den kleinen Finger meiner Frau erkennt der Sensor nicht. Ich habe vieles ausprobiert, auch das weiter am Gelenk auflegen - nutzt alles nichts. Scheint sehr "fingerabhängig" zu sein

heckmic, 2023-02-12:
Bei uns das gleiche. Die Finger meiner Frau sind scheinbar zu klein. Die Finger meines 9 Jähringes Sohnes sind scheibbar groß genug.

sanschaussure, 2024-03-18:
Guten  Morgen,
ich habe auch den Sensor installiert. Das mit den kleinen Fingern funktioniert bei mir einigermaßen wie beschrieben. Vor allem muss der Finger nicht nur an der Spitze aufgelegt werden.
Ein anderes Problem ist bei mir allerdings, dass von einem Kind das Auflegen des Fingers nicht erkannt wird. D.h. am Wakeup-Ausgang (blaues Kabel) kommt kein Trigger an. Habe schon verschiedene Dinge versucht wie andere Schuhe, barfuß, alle Finger durchprobiert ... Bei allen anderen inkl. Kindern funktioniert es gut.
Hatte jemand dieses Problem schon mal?

PinkyFooFoo, 2024-05-25:
Hallo zusammen,

mittlerweile deutlich später habe ich immer noch das Problem, das ich Fehler 6 bekomme (Image mess)
Es funktioniert weiterhin bei meiner einen Tochter nicht. Ich habe zwei unterschiedliche ESPs mit zwei unterschiedlichen Fingerprintsensoren - beide melden bei ihr das gleiche.

Gibts noch die Chance darauf dass man die Anlernvorgänge reduzieren kann über die UI? Das würde ich gerne einmal ausprobieren. 
Allgemein vermute ich hier eher ein Problem mit zu trockener Haut oder ähnlichem als mit der Fingergröße. Die sollte definitiv passen.