frickelzeugs/FingerprintDoorbell #28: Security ESP32 - Flash Readout

joBr99, 2022-11-08:
Hey,
nice that you tought of the attack vector of replaceing the fingerprint sensor. (https://github.com/frickelzeugs/FingerprintDoorbell#pairing-a-new-sensor)

Would like to point out/discuess another potential attack vector with this issue, if the esp32 is accessable from outside it should easy for an attacker to readout the content of the flash chip and simply extract the wifi and mqtt password from the dump.

A potential solution for this could be the Flash Encryption that is possible to setup with ESP32.

https://docs.espressif.com/projects/esp-idf/en/release-v3.2/security/flash-encryption.html

Did you look into this before? 
I might look into it after my fingerprint sensor & the esp has arrived from china.

Thanks a lot for this project, looks way better than anything else I've seen for thoose fingerprint sensors.

Best,
Johannes

frickelzeugs, 2022-11-09:
Hi Johannes,

thanks! You're absolutely right that this is another possible attack vector. At some point during implementation I also thought about this but only briefly read up on the topic of Flash Encryption with the ESP32. But since it seemed quite complex, I didn't pursue the topic any further and limited myself to the pairing for the time being, since the risk of an attack here seems significantly higher. I was also a bit scared by the eFuses, which, once burned, cannot be reset. Bricked devices are then just a matter of time ;-) Also firmware updating doesn't seem to work that easily with activated flash encryption. The best protection against a readout would still be to protect the ESP32 from physical access and to place it e.g. inside the house and only lead the sensor cables through the wall to the outside. But not everyone will want to or be able to do that.

So, you are absolutely welcome to read up on the topic of FlashEncryption and bring in/discuss your ideas or even do a PR!

Regards,
Tobias

joBr99, 2022-11-09:
I actually think that reading out the information stored on the ESP is the bigger attack vector since you only need to connect the usb cable, readout the content and then you are able to do everything remotely. (and getting the strings from the dump shouldn't be very complex) 

For replacing the fingerprint sensor you need to have a replacement one or something that emulates the serial communication of that sensor. (I also didn't not have a look how you implemented the paring, the Datasheet of the R503 Sensor mentions a Handshake Password, however it doesn't look like there is any challenge response involved so it won't help a lot ...)

I also thought of putting the ESP32 inside, especially since I have 5 wires behind my doorbell that I could use, but I guess TTL is not going to work over multiple meters to my fuse box.

joBr99, 2022-11-09:
Proof of Concept:

Dump Flash Content: `esptool.py --baud 921600 --port COM17 read_flash 0x0 0x400000 fw-backup-4M.bin`
Extract printable string from the dump: `strings fw-backup-4M.bin > strings.txt`

Output contains mqtt password, same for the wifi password (wifi password is multiple times in output, for whatever reason):

[image]

Will keep you posted.

joBr99, 2022-11-09:
Encrpytion isn't supported by arduino/platformio.
https://github.com/platformio/platform-espressif32/issues/305

However it looks like it is possible to workaround this by using the bootloader from ESP-IDF.
https://motius.de/insights/secure-iot-prototype-with-arduino-esp32/

joBr99, 2022-11-10:
ported to esp-idf with using arduino as component, didn't yet look into encrpytion

1. Install ESP IDF - choose `release/v4.4 (release branch)` as version in Setup
https://github.com/espressif/vscode-esp-idf-extension/blob/master/docs/tutorial/install.md

2. Clone https://github.com/joBr99/FingerprintDoorbell including submodules.

Looks like it is still working fine.

ChrisLSaar, 2022-11-16:
Hey joBr99

> ~Looks like it is still working fine.~ Sadly it's not really stable.

Have you made any progress here?
What do you mean by not really stable?
I am also interested in this feature and would check it out in the near future. But do not want to do a duplicate work.

Best Christian

joBr99, 2022-11-16:
Hey, 
it compiles you can flash it and it appears to work, but the esp32 is crashing really often. 
Didn't investigate why.

szaiser, 2025-03-05:
> you don't need the Webinterface - Wifi and Mqtt credentials are enough to unlock the door
Even worse, yes. This critical security vulnerability should be mentioned on the main page, I doubt this is known to many users.

joBr99, 2025-03-05:
I solved this by putting the ESP32 inside (luckily I have 4 wires + a "shielding" wire to my doorbell) - between the Fingerprint Reader and the ESP is some security stuff with the pairing.

Even though I'm pretty sure that it is possible to get around it, but you would need to sniff serial communication during a valid finger is on the sensor.