frickelzeugs/FingerprintDoorbell #41: Ring Event with ignoreTouchRing=on

baetzmr, 2022-11-19:
Hallo zusammen. Ich habe hier ein seltsames Verhalten mit der Funktion ignoreTouchRing. Wenn ich die Funktion über mqtt aktiviere bekomme ich das Feedback das ignoreTouchRing=on ist. Das scheint aber bei mir nicht immer sauber zu gehen. Es kommt vor das nach mehrmaligen auflegen des Fingers trotzdem ein Klingelevent ausgelöst wird. Ich erkenne das weil dann auch immer mein Relais für den Gong schaltet. Wenn das passiert blinkt der LED Ring mehrmals Rot und dann schaltet das Relais. Hat jemand eine Idee was ich da tun kann? Danke schon mal

baetzmr, 2022-11-21:
Kann geschlossen werden. Workaround: Einführung einer zusätzlicher Variable hat geholfen. Danke für das schöne Projekt!

nobbi1991, 2022-11-21:
Willst/Kannst du noch erklären, was du geändert hast?

baetzmr, 2022-11-22:
Ich lade später wenn ich wieder daheim bin meine Änderungen hoch. Ist noch ein wenig mehr verändert und auf unser Zuhause angepasst. Bis später...

baetzmr, 2022-11-22:
main.cpp
[image]

[image]

[image]

frickelzeugs, 2022-11-22:
Hi, also das ignoreTouchRing war nie dazu gedacht, das Klingelsignal abzuschalten, lediglich die Sensitivität des Touchs (sinnvoll bei Regen) zu reduzieren, indem die Berührung des Touchrings ignoriert wird und nur der Sensor selbst ausgewertet wird. Ein Auflegen des Fingers auf die schwarze Sensorfläche löst also immer noch die Klingel aus, ist schließlich immer noch eine Fingerprint**Doorbell** ;-)

Das was du willst ist ein anderes Feature, nämlich die Klingelfunktion über MQTT komplett zu deaktivieren.