frickelzeugs/FingerprintDoorbell #56: Anschluss kabelgebundener Gong

RolloD85, 2022-12-21:
Hallo Zusammen,

erst einmal ein riesiges Lob an das Projekt. Ich hab mir gleich die benötigten Teile besorgt und alles umgesetzt. Die Fingerabdruck erkennt funktioniert tadellos und öffnet mit dir Tür.
Leider habe ich aber bei der Klingelfunktion so meine Schwierigkeiten....
Ich benötige diese Möglichkeit, das man den Gong auch über Draht (GPIO 19) auslösen kann. Ich habe schon alle möglichen Klemmversuche unternommen, aber leider bleibt der Gong stumm:(
Ich hoffe mir kann hier jemand schnell helfen. 
Vielen Dank für eine Antwort.
Hier ist der Verdrahtungsplan, meines Gongs. ich hoffe das hilft weiter:
[Verdrahtung Gong.pdf](https://github.com/frickelzeugs/FingerprintDoorbell/files/10276488/Verdrahtung.Gong.pdf)

mickler2021, 2022-12-21:
Hi, den ESP32 wirst Du nicht direkt an den Gong anschließen können. Über den GPIO bekommst du nur ein kurzes Steuersignal von 3V. Dein Gong läuft mit 12V. Daher musst Du wohl ein Relay zwischenschalten. Ich habe dafür folgendes Relais verwendet. DFRobot Digitales 10A-Relaismodul, Digital 10A Relay Module, DFR0473, 5V, 3.3V, für Arduino, Raspberry Pi, BeagleBoard Black, Mbed, Intel Joule https://amzn.eu/d/d4GMZ93

RolloD85, 2022-12-21:
Hallo mickler2021,

danke für deinen Ratschlag und den Link für das Relais. Habe es gleich bestellt:)
Könntest du mir eventuell, nur das ich es auch verstehe, erklären oder zeigen wie du es verdrahtet hast.
Vielen Dank für deine Antwort.

mickler2021, 2022-12-21:
Schau einfach ins ProduktWiki. Dann sollte es Dir klar werden. https://wiki.dfrobot.com/Gravity__Digital_Relay_Module__Arduino_%26_Raspberry_Pi_Compatible__SKU__DFR0473

RolloD85, 2022-12-23:
Ich habe es hinbekommen und alles funktioniert wie es soll. Vielen Dank nochmal.

closed 2022-12-23 by RolloD85