frickelzeugs/FingerprintDoorbell #60: Gingauslösung ohne Fingerabdruck

RolloD85, 2023-01-09:
Hallo zusammen,

ich hab das Problem, das ab und an die Klingel klingelt aber niemand an der Tür war. Im Protokoll wird dann angezeigt, das kein Übereinstimmung gefunden wurde (Code 2).
Ich hab auch schon eingerichtet das bei Wind und Regen der "IgnorTouchRing" auf on ist.
Das Phänomen tritt aber auch bei schönem Wetter auf.
Woran kann das liegen oder was kann ich da noch machen.
Mein Frau ist schon leicht genervt davon 🙈
Vielen Dank für eine Antwort.

frickelzeugs, 2023-01-13:
Abgesehen vom Problem mit den Regentropfen hatte ich bei mir noch keine Fehlauslösungen der Art. Vor allem, wenn du schon ignoreTouchRing permanent auf "on" hast bedeutet das ja, dass der Sensor tatsächlich ein Bild von was auch immer eingescannt hat. Hast du das Metall-Gehäuse auch geerdet? Ansonsten hätte ich auch keine Erklärung und es muss an dem Sensor liegen...

RolloD85, 2023-01-13:
Hallo frickelzeugs,

danke für deine Antwort. Ich muss jetzt mal ganz blöd fragen.....Wie eerde ich da Metallgehäuse am besten?
Ich habe in diesem Zusammenhang auch gleich mal noch eine andere Frage...
Kann man die LED bei IgnorTouchRing auch ausschalten bzw. die Farbe wechseln?
Aktuell leuchtet sie ja immer blau.
Vielen Dank für deine Antwort.

frickelzeugs, 2023-01-13:
Das Sensorgehäuse hat ja hinten eine Kontermutter, da hab ich einfach einen Draht eingeklemmt und mit dem Schutzleiter (PE) verbunden.

Die LED kann man mit meiner Version aktuell nicht abschalten oder die Farbe einstellen. Es gibt da aber einen Pull Request eines anderen Users, der das programmiert hat. Den müsste ich bei Gelegenheit mal noch in die offizielle Version mergen.

RolloD85, 2023-01-13:
Da werde ich das mit dem erden mal ausprobieren und hoffe das es daran gelegen hat. 

Ich kenne dieses Pull Request bereits und nutze es auch. Ich habe dort die Led-Sequenz auf aus. Wenn ich aber jetzt IgnorTouchRing auf on schalte geht der Led-Ring wieder an. Ist das ein normales Verhalten?

PinkyFooFoo, 2023-01-14:
Tag zusammen,

ich habe ähnliches bemerkt. War bei mir diverse male eine Spinne, etwa 1mm groß die auf dem Fingerabdrucksensor umher gelatscht ist. Der Touchring war ebenfalls auf ignore gestellt, es war also der Fingerabdruck-Sensor selbst der die Spinne als "Bild" erkannt hat.

Phoenxi131178, 2023-02-10:
Moin, ich habe nun herausgefunden, das wenn bei mir ein mqtt Gerät ein on schickt die Klingel ebenfalls auslöst.
Obwohl das ON über ein anderes Topic kommt. Ich etwas verwundert und suche selber noch noch einer Lösung.

mkrasselt1, 2023-02-16:
schau mal lieber wie die logic deiner klingel aussieht.
Kann es sein, das die Klingel einen Platzhalter verwendet bei Ihrer Automatisierung ?

Phoenxi131178, 2023-02-16:
> schau mal lieber wie die logic deiner klingel aussieht. Kann es sein, das die Klingel einen Platzhalter verwendet bei Ihrer Automatisierung ?

Ich benutze Node Red für die Automatisierung und darin die mqtt in Node.
Die Klingel gibt ein Mqtt Signal aus genau wie es sein soll. Leider halt auch wenn andere Geräte per mqtt geschaltet werden. Dies gibt die Klingel aus:
topic: "fingerprintDoorbell/ring"
payload: "on"
qos: 0
retain: false
_msgid: "554ef5671899cc4d"
sollte soweit richtig sein.

Möglich ist auch das ich den Sensor falsch an den ESP gelötet habe aber es geht alles außer, dass die Klingel auslöst wenn ein anderes Gerät über Mqtt feuert, egal welches Topic. Es ist aber nicht direkt reproduzierbar.
Ich habe nun einfach einen zweiten Sensor (BW) in Node Red dazwischen gepackt. Wenn beide on sind geht die Klingel somit habe ich das Problem für mich umgangen aber nicht gelöst.
Danke für die Hilfe aber ich denke es ist kein Fehler oder Problem des Projektinhabers hier.
Falls ich da noch mal tiefer reingehe werde ich das ganze ins Community Forum stellen.
Vorher muss ich aber meine Bastellei nochmal überprüfen ;)