frickelzeugs/FingerprintDoorbell #61: Receiving a second incorrect Mqtt message

PregLizZz, 2023-01-09:
Hi,
first of all, thanks a lot for this amazing project. I have the following problem:
When the finger is placed on the sensor and removed I always receive a the correct mqtt messages, with
lastLogMessages, ring, matchId, matchName, matchConfidence 
always filled with the correct values.
3 seconds later I always receive another set of messages always not recognized and always without lastLogMessages so only:
ring, matchId, matchName, matchConfidence
In the grey div of http://fingerprintdoorbell/ I only see the correct message not the unrecognized one 3 seconds later. The problem seems to be only in mqtt.
I've already tried to factory reset but no luck. Maybe I'm overlooking something and someone can help me out.
Thanks for your help.

frickelzeugs, 2023-01-13:
Hi,

thats the designed behaviour and also mentioned in the documentation:
<html><body>

MQTT Topic | Action | Values
-- | -- | --
fingerprintDoorbell/ring | publish | "off" by default, on a ring event switching to "on" for 1s
fingerprintDoorbell/matchId | publish | "-1" by default, if a match was found the value holds the matching id (e.g. "27") **for 3s**
fingerprintDoorbell/matchName | publish | "" by default, if a match was found the value holds the matching name **for 3s**
fingerprintDoorbell/matchConfidence | publish | "" by default, if a match was found the value holds the conficence (number between "1" and "400", 1=low, 400=very high) **for 3s**
fingerprintDoorbell/ignoreTouchRing | subscribe | read by FingerprintDoorbell and enables/disables the touch ring (see FAQ below for details)

</body>
</html>

You have to evaluate the payload of the mqtt topic, not only do stuff when the topic was sent.

joBr99, 2023-01-31:
As mqtt is usally more event based and it's easier to work with if you just have all values available I've personally added a new topic `/match` which is going to send all information as json in one message.

```
        mqttClient.publish((String(mqttRootTopic) + "/match").c_str(), String("{\"matchId\": " + String(match.matchId) + ",\"matchName\": \"" + match.matchName + "\",\"matchConfidence\": " + String(match.matchConfidence) + "}").c_str());

```