frickelzeugs/FingerprintDoorbell #64: Secirity Issue

Stingray79, 2023-02-05:
Folgende Fehlermeldung wird mir angezeigt:

Security issue! Match was not sent by MQTT because of invalid sensor pairing! This could potentially be an attack! If the sensor is new or has been replaced by you do a (re)pairing in settings page.

Was kann ich dagegen tun?

RolloD85, 2023-02-11:
Hallo,

bei mir ist das gleiche Problem gestern auch aufgetreten.
Gibt es hierzu schon eine Lösung?
Vielen Dank für eine Antwort.

Gruß Ronny

frickelzeugs, 2023-02-11:
Auch mit den selben Fehlercodes wie Stingray79 und ein erneutes Pairing ist nicht mehr möglich? Einfach so aus heiterem Himmel oder hast du vorher irgendwas anders gemacht? Schon probiert alles stromlos zu machen? Evtl. hat sich der Sensor aufgehängt?

Die Fehlercodes oben passen eigentlich auch gar nicht zum Pairing-Prozess:
`#define FINGERPRINT_FEATUREFAIL                                                \
  0x07 //!< Failed to generate character file due to the lack of character point
       //!< or small fingerprint image
#define FINGERPRINT_NOTFOUND 0x09 //!< Failed to find matching finger`

Und eine confidence >1000 hab ich auch noch nie gesehen, alles sehr seltsam...

RolloD85, 2023-02-11:
Stromlos habe ich schon probiert und ein neues Pairing schlägt leider fehl.
Ich habe nichts anders gemacht als sonst. Das einzigste was war, das wir einen kurzen Stromausfall hatten. Hier auch nochmal ein Screenshot...
[image]

RolloD85, 2023-02-11:
Nach dem ich mehrfach versucht habe neu zu Paaren, kommt jetzt nach einem Neustart folgende Fehlermeldung:

Security issue! Pairing with sensor is invalid. This could potentially be an attack! If the sensor is new or has been replaced by you do a (re)pairing in settings page. MQTT messages regarding matching fingerprints will not been sent until pairing is valid again.

Vielen Dank das du dir das mal ansiehst.

Stingray79, 2023-02-11:
Ich habe meinen noch gar nicht in Betrieb bekommen, habe seit Anfang an, das Problem

Tar00, 2024-04-09:
Ich habe die gleiche Fehlermeldung nach ca
 24 Stunden Betrieb bekommen. Gibt es dazu schon eine Lösung?