frickelzeugs/FingerprintDoorbell #67: fingerprint detection fast - notification and mqtt slow

Zappelphilipp, 2023-02-26:
Hi,

Thank you for the good work. I am coming from this project: https://github.com/vinz486/fingerprint-r503-mqtt

and basically everything is much more polished here so it was easy to deside, that i have to switch ;)

i only do have one problem: even thouch my finger is recognized very fast on the sensor side (the blue light comes about 1s after putting my finger on it), the mqtt messages and also the notifications at the top of the web gui are taking about 3-5 seconds to appear. long after the finger was recognized and the LED's switches color.

can this problem mitigated somehow?

mkrasselt1, 2023-03-01:
what is the reception for your device and is there any interference source nerby - does the problem occur anywhere ?

Zappelphilipp, 2023-03-03:
yes, the problem does occur on two different places with two different wifi-AP-vendors. The reception is basically 100% because i tested right next to the AP.

The web GUI is also super responsive and opens instantly, only the fingerprint/ring detection is slow.

weirdly one time the response was instant but i don't know why. i looked through the code a bit and it seems like the program goes into a different mode during "Web GUI use". maybe then it's slower?

Zappelphilipp, 2023-03-18:
Update: maybe it has something to do with the NTP Sync because the environment is airgapped and on the console i can see on every finger detection that it tries to get the NTP time which obviously fails and maybe this takes a few seconds?

can someone confirm this? i dit not find anything in the code though...

closed 2023-03-22 by Zappelphilipp