frickelzeugs/FingerprintDoorbell #71: Enrollment geht nicht

net09apphome, 2023-04-13:
Hallo !

tolles Projekt ! Habe das Problem das das enrollment nich funktioniert! Der Scanner blinkt purple aber sonst passiert leider nichts !

wenn ich den scanner starte, liest er den Finger und zeigt rot an ! Was ja aich richtig ist da er den finger nicht kennt
No match found Code 02

Danke im vorraus für die Hilfe

sluhura, 2023-06-17:
Hallo,
ich hatte auch das Problem, Ursache war ein schlechte Stromversorgung (zu viel Restwelligkeit)
br

net09apphome, 2023-06-27:
> kann man das als gelöst betrachten oder hat @net09apphome  trotz perfekter stromversorgung noch probleme ?

HalloJa es ist gelöst!Danke

andy0172, 2024-11-21:
Hallo, auch von meiner Seite vielen Danke für das tolle Projekt. Ich habe auch das Problem, dass das Enrollment nicht geht. Die Aufforderungen zum Finger heben kommen alle nacheinander und dann Direkt Error Code 2
[2024-11-21 10:19:09 UTC]: Enrollment for id #1 started. We need to scan your finger 5 times until enrollment is completed.
[2024-11-21 10:19:09 UTC]: Take #1 (place your finger on the sensor until led ring stops flashing, then remove it).
Taking image sample 1: taken, converted[2024-11-21 10:19:09 UTC]: Take #2 (place your finger on the sensor until led ring stops flashing, then remove it).
Taking image sample 2: taken, converted[2024-11-21 10:19:09 UTC]: Take #3 (place your finger on the sensor until led ring stops flashing, then remove it).
Taking image sample 3: taken, converted[2024-11-21 10:19:09 UTC]: Take #4 (place your finger on the sensor until led ring stops flashing, then remove it).
Taking image sample 4: taken, converted[2024-11-21 10:19:09 UTC]: Take #5 (place your finger on the sensor until led ring stops flashing, then remove it).
Taking image sample 5: taken, Unknown error[2024-11-21 10:19:09 UTC]: Enrollment failed. (Code 2)

Hab es jetzt an einem eigenen Netzteil mit 2A, also daran sollte es nicht liegen. 

Beim Starten kommt auch immer dieser Fehler, egal wie oft ich alle Fingerprints lösche: 
 Warning: Fingerprint count mismatch! 4 fingerprints stored on sensor, but we are aware of 0 fingerprints.

Vielleicht hat ja noch jemand eine Idee, das Pairing klappt erfolgreich. Ich hab das Gefühl er erkennt nicht wirklich ob ein Finger aufliegt, obwohl er darauf reagiert mit der Meldung und startet wieder im AP Modus:
[2024-11-21 10:11:31 UTC]: Match Found: 10 - @empty with confidence of 0
[2024-11-21 10:11:31 UTC]: Security issue! Match was not sent by MQTT because of invalid sensor pairing! This could potentially be an attack! If the sensor is new or has been replaced by you do a (re)pairing in settings page.
Unknown error
[2024-11-21 10:11:34 UTC]: ScanResult Error (Code 9)
Guru Meditation Error: Core  1 panic'ed (LoadStoreError). Exception was unhandled.
Core 1 register dump:
PC      : 0x4018a185  PS      : 0x00060830  A0      : 0x800ebe8d  A1      : 0x3ffb1c40  
A2      : 0x4000c468  A3      : 0x3ffb1c12  A4      : 0x000003e8  A5      : 0x0000000b  
A6      : 0x00000000  A7      : 0x00000008  A8      : 0x35392539  A9      : 0x00000003  
A10     : 0x0000000c  A11     : 0x00000043  A12     : 0x00000005  A13     : 0x3ffb1c20  
A14     : 0x00000000  A15     : 0xef010003  SAR     : 0x00000008  EXCCAUSE: 0x00000003  
EXCVADDR: 0x4000c46e  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

ELF file SHA256: 0000000000000000

Backtrace: 0x4018a185:0x3ffb1c40 0x400ebe8a:0x3ffb1c60 0x400d25e5:0x3ffb1c80 0x400d625a:0x3ffb1ce0 0x400d7509:0x3ffb1f30 0x400ed951:0x3ffb1fb0 0x40089a7e:0x3ffb1fd0

Rebooting...