frickelzeugs/FingerprintDoorbell #72: MQTT Problem

Phoenxi131178, 2023-04-23:
Meine Türklingel reagiert auf andere MQTT Nachrichten. Ich kann es zum Teil reproduzieren. Hat jemand ein ähnliches Problem und konnte es lösen? Ich bin um jeden Ratschlag dankbar.

My doorbell reacts to other MQTT messages. I can partly reproduce it. Has anyone had a similar problem and was able to solve it? I'm grateful for any advice.

thanks..

HansPeterMeier, 2023-04-25:
Bin gerade dabei auszuloten, was ich auf den ESP packe.
Ich hatte bisher einen Fork genommen von https://github.com/TheMaskedDeveloper/FingerprintDoorbell , da ich hier den Port und die LED´s besser ansteuern konnte.
Aber ich habe auch festgestellt, das beim Fingerscannen ab und zu ein unmatch dazwischen war, das würde dann bei mir auch die Klingel auslösen. Ich bin nicht der Programmierer, aber ich muss mal gucken ob man das timing irgendwie ändern könnte.
Schade das der Entwickler von diesen tollen Sketch nicht mehr aktiv ist.
Gruß

Phoenxi131178, 2023-04-25:
Moin, ok aber leider hat das nix mit meinem MQTT Problem zu tun.
Liebe Grüße Jan

HansPeterMeier, 2023-04-25:
Dann hab ich dich wohl falsch verstanden.
aber du schaltet mit dem ESP ne Relais, wenn ein Finger erkannt wurde mit "unmatch" um damit ein Klingel auszulösen?
Gruß

Phoenxi131178, 2023-04-25:
Ja das ist richtig. Aber meine Klingel löst auch aus wenn per MQTT ein anderes Gerät zum Beispiel meine Jalousie runtefährt.
Die Klingel so funktioniert super wenn ich den Finger benutze. Ich hab nur das Problem das Die Klingel auch ein MQTT Signal sendet wenn ein anderes Gerät angesteuert wird.
Obwohl es völlig andere Topics hat.
Gruß Jan

HansPeterMeier, 2023-04-25:
Das ist wirklich schon ein wenig merkwürdig. Sorry, da kann ich leider nicht helfen.

Phoenxi131178, 2023-04-25:
Kein Problem, hab schon geschaut ob ich den Kram falsch zusammen gelötet habe aber es ist alle richtig verkabelt.
Es funktioniert auch super, bis auf die Auslösung von anderen Mqttgeräten.
Ich hab keine Ahnung. Hab schon wer weiss was ausprobiert.
naja trotzdem danke

Phoenxi131178, 2023-04-26:
Ok ich konnte es nun etwas eingrenzen.
Hab die Klingel 1 zu 1 nachgebaut nochmal.
Es läuft alles bis auf das Mqtt Phanomän.

1) Strom Quelle getauscht.
2) Mqtt mehrmals angepasst.
3) Node Red, automation.yaml ausprobiert.

Am Ende kann ich sagen. Wenn ein Gerät bei mir per Mqtt auslöst wird unregelmäßig die Türklingel mit ausgelöst.
Ich hab jetzt einen zweiten Sensor (Bewegungsmelder) hinzugefügt damit die Klingel nicht auslöst wenn niemand vor der Tür steht.

ChrisLSaar, 2023-04-26:
@Phoenxi131178 : Schau mal deine ClientIDs an mit denen du die Teilnehmer am MQTT Broker anmeldest.  Ich hatte mal ein ähnliches Problem, hier war eine Client-ID doppelt vergeben. Normalerweise verweigert der Broker dann einen Connect, in meinem Fall sind aber auch sehr merkwürdige Probleme aufgetreten.

Phoenxi131178, 2023-04-26:
Danke das werde ich machen. Vielen Dank

Phoenxi131178, 2023-04-26:
@ChrisLSaar ClienIDs sind alle verschieden.  Aber danke für den Tipp

Phoenxi131178, 2023-05-04:
Kurz zur Info ich habe wohl das Problem lösen können.
Ich hatte für die Türklingel und die anderen Module die über Mqtt senden, den gleichen User benutzt.
Nachdem ich das nun geändert habe scheint es zu laufen.

closed 2023-05-04 by Phoenxi131178