frickelzeugs/FingerprintDoorbell #82: HomeAssistant integration

evilmumi, 2023-11-27:
Hi,

can anyone help to integrate to Homeassistant

evilmumi, 2023-12-07:
> I have add this to my .yaml in HomeAssistant
> 
> ```
> sensor:
> #FingerprintDoorbell  
>   - state_topic: "/SmartHome/System/FingerprintDoorbell/matchName"
>     name: "FingerprintDoorbell matchName"
>   - state_topic: "/SmartHome/System/FingerprintDoorbell/matchId"
>     name: "FingerprintDoorbell matchId"
>   - state_topic: "/SmartHome/System/FingerprintDoorbell/ring"
>     name: "FingerprintDoorbell ring"
>         
> #Fingerprint Doorbell
> switch:
>   - unique_id: FingerprintDoorbell-IgnoreTouchRing
>     command_topic: /SmartHome/System/FingerprintDoorbell/ignoreTouchRing
>     payload_on: "off"
>     payload_off: "on"
>     name: "FingerprintDoorbell IgnoreTouchRing"
> ```

Thanks a lot.
I just added an automation but it calls every time.
How did you configure some?
```

alias: FingerprintDoorbell Name
description: ""
trigger:
  - platform: state
    entity_id:
      - sensor.fingerprint_name
    not_to: ""
condition:
  - condition: not
    conditions:
      - condition: state
        entity_id: sensor.fingerprint_name
        state: unavailable
      - condition: state
        entity_id: sensor.fingerprint_name
        state: unknown
action:
  - service: notify.alexa_media_room_dot
    data:
      message: Hello {{states('sensor.fingerprint_name')}} !
mode: single

```

janboesebeck, 2024-10-19:
Brauche ich dann nicht ein entsprechendes Verzeichnis dafür ? Wenn ich den Code kopieren kommt bei mir der error "platform"

evilmumi, 2024-10-19:
das liegt vermutlich an der yaml änderung von HA - lies dich mal schlau: in den changelogs

janboesebeck, 2024-10-19:
sind die Verzeichnisse denn bei dir ebenfalls andere oder wie ist es bei dir realisiert ?

evilmumi, 2024-10-20:
Meinst du mit Verzeichnissen den Namen der mqtt topics?

janboesebeck, 2024-10-20:
Tatsächlich habe ich es durch ausprobieren geschafft die Entitäten hinzuzufügen. Wie ich das verstanden habe sollte jedoch die Übertragung irgendwo prüfbar sein. Die ganzen Entitäten sind auf unbekannt. Auch wenn ein Finger erkannt wird oder nicht. Im MQTT Mosquitto Log ist auch nur das einloggen nach dem Neustart des ESP zu sehen. Ne idee was ich falsch mache? 
Bin echt am verzweifeln bei dem Projekt