frickelzeugs/FingerprintDoorbell #84: Security: DNS Request

dbtec00, 2023-12-09:
Hi, 

ich habe PiHole im Einsatz und sehe, dass hier kontinuierlich (alle 20s) versucht wird eine fehlerhafte Domain vom Controller aufzulösen. Ist das bekannt? 

Grundsätzlich möchte ich nicht, dass hier irgendeine Kommunikation stattfindet von einem Gerät, welches meinen Fingerabdruck gespeichert hat (wurde auch bereits geblockt). Aber warum diese Anfragen?

[image]

Grüße
db

Noschvie, 2023-12-09:
Oh... eben reproduziert, was soll das bzw. von wo im Code kommt das?

Noschvie, 2023-12-09:
@dbtec00 hast du probiert, den Code selber zu compilieren und laufen zu lassen, ob reproduzierbar?

ChrisLSaar, 2023-12-09:
Hatte diese Anfragen auch. Sie entstehen durch den NTP Request wenn etwas geloggt werden sollte.
Hier wird die Zeit vom parametrierten NTP Server abgerufen und vor die LogMessage gesetzt.
Trägt man den Zeitserver in der Konfiguration neu ein und speichert die Konfiguration funktioniert es für eine gewisse Zeit wieder.
Irgendwann scheint aber intern irgend ein Teil vom Speicher überschrieben zu werden und es wird eine kryptische Domäne am DNS Server angefragt.
Ich habe ziemlich lange danach gesucht, konnte die Ursache aber auch nicht finden.
Habe die getTimestampString Methode stillgelegt und suche bei Gelegenheit weiter nach dem Problem.

```
String getTimestampString(){
  struct tm timeinfo;
  return "no time"; //Temporäre bis NTP wieder geht
}
```

dbtec00, 2023-12-10:
> @dbtec00 hast du probiert, den Code selber zu compilieren und laufen zu lassen, ob reproduzierbar?

Nein, nur direkt über die Bin.

dbtec00, 2023-12-10:
> Hatte diese Anfragen auch. Sie entstehen durch den NTP Request wenn etwas geloggt werden sollte. Hier wird die Zeit vom parametrierten NTP Server abgerufen und vor die LogMessage gesetzt. Trägt man den Zeitserver in der Konfiguration neu ein und speichert die Konfiguration funktioniert es für eine gewisse Zeit wieder. Irgendwann scheint aber intern irgend ein Teil vom Speicher überschrieben zu werden und es wird eine kryptische Domäne am DNS Server angefragt. Ich habe ziemlich lange danach gesucht, konnte die Ursache aber auch nicht finden. Habe die getTimestampString Methode stillgelegt und suche bei Gelegenheit weiter nach dem Problem.
> 
> ```
> String getTimestampString(){
>   struct tm timeinfo;
>   return "no time"; //Temporäre bis NTP wieder geht
> }
> ```
> 
> Ich weiß, nicht die beste Lösung aber wollte auch bloß schnell die ständigen Abfragen verhindern.
> 
> P.S.: Die Methode "getLocalTime" fragt die Zeit im System ab. Diese ist in der Lib "esp32-hal-time". Hier wird dann aus der Lib raus auf den parametrierten NTP Server zugegriffen.

Getestet und bestätigt! Neues setzen des NTP Servers behebt das Problem! 

[image]

Noschvie, 2023-12-10:
```
  char buffer[25];
  strftime(buffer,sizeof(buffer),"%Y-%m-%d %H:%M:%S %Z", &timeinfo);
  String datetime = String(buffer);
  return datetime;

```

kann morgen jemanden bitten, diese Zeilen Code zu prüfen.

ChrisLSaar, 2023-12-10:
@Noschvie : Auf was genau willst du hinaus? 
Das ist die Kovertierung des aus "getLocalTime(&timeinfo)" ausgegebenen Datentyp in ein lesbares Format.
Die Methode trägt ihr Ergebnis in den Char-Buffer "buffer" ein.
Mit sizeof(buffer) wird sichergestellt dass die Speichergröße des Buffers nicht überschritten wird.

[C++ Reference](https://en.cppreference.com/w/c/chrono/strftime)

Danach noch die Konvertierung als String.
Wenn ich nichts übersehen habe sollte das sauber sein.

Interessant wäre es raus zu finden ab wann die Abfrage unsauber wird.
Bestimmtes Ereignis wie nicht erkannter Fingerabdruck, nach einer bestimmten Zeit, nach gewisser Anzahl von Abfragen, ...

Würde der Sache gerne näher auf den Grund gehen, habe aber nur einen Fingerabdruckleser der bei mir "produktiv" läuft.
Da kann ich nicht rum basteln. Einen neuen kaufen nur zum Debuggen ist mir die Sache jetzt nicht wert.

Wenn aber jemand näheres raus findet schaue ich mir es gerne an.

Noschvie, 2023-12-10:
Irgendwo scheint es einen Speicherüberlauf zu geben, nur ich sehe den nicht mangels Erfahrung mit C/C++

ChrisLSaar, 2023-12-10:
@Noschvie : Da hast du vermutlich recht mit dem Speicherüberlauf, der kann aber so ziemlich überall sein. Ohne richtiges debuggen zur Laufzeit finde ich den aber auch auch nicht. Da reichen meine C++ Kenntnisse auch nicht aus.

An der Stelle die du vermutest ...

```
char buffer[25];
strftime(buffer,sizeof(buffer),"%Y-%m-%d %H:%M:%S %Z", &timeinfo);

```
Die Methode strftime mir den Parametern sollte eine Zeitstempel erzeugen der folgendermaßen aussieht...

2023-12-10 16:25:35 MEZ bzw. (MESZ)

Sind also 23 (24) Zeichen die in den Buffer geschrieben werden.
Dann braucht man zur Umwandlung in einen String noch die null-character Terminierung, also nochmals ein Char mehr.
Damit ist man noch bei 25 Zeichen zumindest in hiesigen Gefilden.
Lebt man jetzt in Novosibirsk, dann passt es in der Tat nicht mehr, zumindest im Sommer nicht (NOVST).

Ich mache den Buffer mal größer und versuche mal was bei raus kommt.

Noschvie, 2023-12-10:
Verwendest du VS Code? Damit kann man eine statistische Code Analyse machen, oder? Vielleicht findet die was.

dbtec00, 2024-04-19:
Hi, hat noch jemand etwas rausgefunden? Ich bin leider kein Programmierer aber das ist schon ziemlich nervig !