frickelzeugs/FingerprintDoorbell #87: Connection 38 pin board

xxbyxx, 2024-04-11:
Hi, 

weiß jemand wie die Pin-Belegung am 38pin esp32 aussieht? 
Ich habe hin und her probiert aber der R503 sensor scheint nicht angesprochen zu werden (kein LED Ring und auch keine Reaktion auf das Anlernen neuer Finder im web UI)

p.s. mein R503 hat ein braunes statt grünes Kabel - gibt es da grundsätzlich unterschiedliche Belegung der Kabel in den gängigen Sensoren oder ist das tatsächlich einfach nur eine andere Farbwahl?

Vielen Dank vorab 

[image]

xxbyxx, 2024-04-11:
I fixed it

closed 2024-04-11 by xxbyxx

Noschvie, 2024-04-12:
Wie sieht deine Lösung aus?
Für jene, die hier landen und nach einer Antwort suchen...

xxbyxx, 2024-04-12:
Klar - sorry ich hatte gestern kaum Zeit für eine ausführlichere Antwort. 
Im Grunde muss man nur das GPIO Belegungsschema konsequent auf die 38-Pin-Variante übertragen
Jetzt funktioniert auch die Kommunikation mit dem IO broker

[image]