frickelzeugs/FingerprintDoorbell #88: Fingerprint reader only works with shoes on

DUSAG0211, 2024-05-04:
Hey,
not an electrician but apparently the reader only detects presence of my finger if i have shoes on.
If in socks and touching the ground in front of my house the reader does not work.
If i climb on my wooden bench it works again.

Can someone more literate in electric stuff share what i would need to change in my system?

I am powering the device with approx 5.6v from our doorbell wire.
The power comes from a 5v dc adapter in our power box (Meanwell Hutschienen-Adapter)

geronet1, 2024-07-01:
L & N am Netzteil vertauschen :-)

geronet1, 2024-07-01:
Kann gut sein daß da ein Ableitstrom von dem Sensor (GND) über dich auf Erde läuft und er den Finger dadurch nicht erkennt.
Evtl. auch anderes Netzteil probieren je nachdem ob (-) geerdet ist oder nicht

xxbyxx, 2024-10-20:
Hi, selbes Problem hier. Liegt bei mir definitiv am Hutschienen-Netzteil (Mean Well HDR-15-5). Ich habe testweise ein Steckernetzteil vom Samsung Smartphone benutzt und damit funktioniert es ohne Probleme. 
Woran liegt das genau?

 Hast du eine Kaufempfehlung
 für ein Hutschienen-Netzteil mit dem das nicht passiert?

DUSAG0211, 2024-11-03:
> Kann gut sein daß da ein Ableitstrom von dem Sensor (GND) über dich auf Erde läuft und er den Finger dadurch nicht erkennt. Evtl. auch anderes Netzteil probieren je nachdem ob (-) geerdet ist oder nicht

Schliesse mich der Frage an: welches Hutschienennetzteil sollte man probieren? Habe auch ein Meanwell hdr im Einsatz

ThomaxK, 2024-11-22:
Ich hatte das gleiche Problem. Mit einem Handynetzteil direkt am ESP ging es problemlos, mit dem Netzteil der Installation ging es nicht mehr.
Das Problem: Der Sensor braucht für die Fingerdruckerkennung eine GND-Verbindung. Die meisten Niederspannungsnetzeile sind jedoch prinzipiell für einen SELV Stromkreis ausgelegt, bei dem der Sekundärstromkreis nicht geerdet ist. Baut man den Niederspannungsstromkreis auf PELV um, so geht der Fingerabdrucksensor einwandfrei - zumindest bei mir.
Dazu muss einfach der Minuspol der Sekundärseite mit dem Schutzleiter verbunden werden.
[image]

xxbyxx, 2024-11-23:
Danke für die Info! 
Ich verstehe nicht ganz wieso es dann mit dem Handynetzteil klappt. Die Stecker haben ja keinen Schutzleiterkontakt, daher ist die Masse(-) bei einem Steckernetzteil doch nie geerdet oder?

DUSAG0211, 2024-11-26:
I can confirm that changing to a pelv setup has solved this issue. Even in socks or barefoot i get instant detection

closed 2024-11-26 by DUSAG0211