frickelzeugs/FingerprintDoorbell #89: Option to disable Ring-Light at night?

Freezy1982, 2024-07-11:
Awesome Project and Idea - working very well for me at home. Thanks for all the effort at this project and the good tutorial ;-) 
I was wondering, if there could be an option to disable the blue indication light during night times (or permanently). In my use-case, I don't want the Fingerprint-Reader to be a "blue-breathing-eye-catcher" at night, as I am using it at and for my garage-door only and I prefer as it would be more hidden...
Any idea if this is possible to be implemented? Maybe same as the TouchRing on/off by MQTT-Message?

mickler2021, 2024-07-11:
Have a look at my fork, I have realized something similar. I can deactivate the bell via MQTT and deactivate the light from the LED ring in the process.

xxbyxx, 2024-07-11:
> Have a look at my fork, I have realized something similar. I can deactivate the bell via MQTT and deactivate the light from the LED ring in the process.

I can confirm it’s working amazing.

Freezy1982, 2024-07-12:
You guys are awesome!
Your update, mickler, is working very nice. I directly changed the colors to my preference as well (luckily I got the new version of the sensor).
This Fingerprint-Doorbell is making my day (and night) now :-D

DUSAG0211, 2024-07-21:
> Have a look at my fork, I have realized something similar. I can deactivate the bell via MQTT and deactivate the light from the LED ring in the process.

I am still on maskeddeveloper fork but noticed that the ring comes back on as long as the touchring detection  is turned off. Is it the same in your build?

mickler2021, 2024-07-21:
The ignoretouchring function works as expected. I have not noticed that the touch ring is reactivated after a while.

DUSAG0211, 2024-07-21:
Oh sorry i meant that the led color comes back on. 
so i have disabled it by default through the additional settings introduced by maskeddeveloper. However if touchring is disabled the led comes back on (standard blue color, always on)

mickler2021, 2024-07-21:
The blue color is hard-coded in the function for the ignoretouchring by maskeddeveloper. When irgnoretouchring is active, the LED ring turns blue.

DUSAG0211, 2024-07-21:
Alright, seems i missed that. I will look through the code again. Thanks for the pointer.

Rik74, 2025-08-19:
got it now!
did not know what i have done in an other way, but now i have the additional options
Thanks!