frickelzeugs/FingerprintDoorbell #90: Kein Zugriff auf Webinterface

RolloD85, 2024-07-12:
Hallo Zusammen,

seit einem Update habe ich keinen Zugriff mehr auf das Webiterface. Die Doorbell an sich funktioniert aber und sendet auch Daten an MQTT. Woran kann das liegen? Habe ich etwas falsch gemacht?
Vielen Dank für eine Antwort.

nevries, 2024-08-20:
Had the same problem when building the firmware myself (with platformio). Turns out I need to build and flash the SPIFF image as well. Maybe when you updated that wasn't done / failed?
You can check by trying to open the web ui and check the browser development tab under network for the return code. If it's 404 instead of something in the 2xx, then most likely the SPIFF is not loaded correctly (means, the webserver on the device won't find the html files to return).

RolloD85, 2024-08-22:
Ich habe jetzt einfach alles neu aufgespielt und alles funktioniert wieder. Aber trotzdem danke für den Tipp.

closed 2024-08-22 by RolloD85