frickelzeugs/FingerprintDoorbell #91: Fingerabdrücke vom Sensor laden

RolloD85, 2024-08-22:
Hallo Zusammen, 

ich habe folgendes Problem....
Ich habe mir einen neuen ESP32 D1 Mini zugelegt und die Software der Fingerprintdoorbell aufgespielt. Danach habe ich meinen alten Sensor wieder angeschlossen, alles konfiguriert und es klappt wieder alles. In der Weboberfläche zeigt es mir keine gespeicherten Fingerabdrücke an, wenn ich den Sensor aber betätige sehe ich im Log das meine Abdrücke alle noch vorhanden sind und auch richtig erkannt werden. 
Gibt es eine Möglichkeit diese Abdrücke, die vermutlich noch auf dem Sensor hinterlegt sind, über die Weboberfläche zu verwalten bzw. diese dort wieder sichtbar zu machen?
Vielen Dank für eine Antwort.

Gruß RolloD85

Panoru84, 2024-10-17:
Wenn man den hier aus den Readme diesen Teil liest:

`Managing fingerprints
The sensor has the capacity for storing up to 200 fingerprints. Theses memory slots are used as ID together with a name to increase human readability. To enroll new fingerprints enter a ID and name (optional) in the "Add/Replace fingerprint" section and click "Start enrollment". Now the system asks you to place and lift your finger to the sensor for 5 times. The 5 passes of scanning helps the sensor to improve its recognition rate. Don't try to vary your placing/position too much, because the enrollment process may fail if the 5 preceeding scans differ too much from each other and cannot be combined to one fingerprint template.`

Müsste auf den Sensor die ID und der Name gespeichert sein.
Ich hätte jetzt auch gedacht die werden dann wieder übertragen, wenn du einen neuen ESP nimmst.

Hast du das ganze nochmal gepairt? Oder werden die gespeicherten Abdrücke dann gelöscht?

`Pairing a new Sensor
For security reasons the ESP32 and Sensor will be coupled together, so if the sensor is replaced (e.g. an attackers connects his own sensor to the ESP32 with his fingerprints on it) this will be detected. In this case the pairing will be marked as broken and no further match events are sent by MQTT from now on (even if you connect the old sensor again). But keep calm, the doorbell function will still continue to work and ring events are sent by MQTT so you don't miss your long awaited package delivery. You'll see an error message in the log window that requests you to renew the pairing. If the sensor replacement was done by yourself or no attack took place please choose the option "Pairing a new Sensor" to pair the sensor with the ESP32.`

RolloD85, 2024-10-23:
Bei mir hat es letztendlich den Scanner zerschossen und somit musste ich mir einen neuen zulegen und alles neu anlernen.