frickelzeugs/FingerprintDoorbell #95: Cabling issue?

xxbyxx, 2024-10-14:
Hi, 

some days ago I changed position of the ESP32. Forehand it was directly attached to the sensor, but outside. Since I was afraid about winter weather conditions, I moved it to the switchboard cabinet. Therfore I used an existing in-wall ethernet cable to connect sensor and ESP32. Cable distance is approx. 15m. 

Since that change I experienced several issues. E.g. sensor triggers ring via MQTT if I close motor driven covers randomly or sometimes there is not any touch sensitivity. 
The issues does not occur if "fingerprintDoorbellForkMini/ignoreTouchRing" is on! So it might has to do something with the touchring? 

Did anybody experienced comparable issues and might come with a hint? Do you think cable distance is to long or is ist more likley that there is any EMS disturbance coming from 230V cables next to the ethernet cable?

KR 
Florian

geronet1, 2024-10-14:
That's too long. If you need that distance use RS485 tranceivers on both
sides to get a differential signal on the cable.

xxbyxx, 2024-10-14:
Thanks mate. 
Unfourtunatelly I am not familiar to that interfaces. Is it a connector on both sites which makes the communication more robust to interferences? 
Can you post a link to the right components? I looked at google but not quite sure what to order… 
Is it plug and play then? 
Thanks in advance

geronet1, 2024-10-14:
Because i want to extend the distance between the sensor and the ESP32 as well to around six meters, we first must check if the serial stream of the sensor is full oder half-duplex. If it is only half-duplex, we need only two wires for tx and rx (and GND), one for the finger detection signal and one for the power supply.

If the power supply voltage is 5 V, we must use some sort of voltage regulator to get 3.3 V for the sensor, the rx line should be connected to the interface ic over a voltage divider.

Important is the ability of autodirection-control, look at the circuit on page 7:
https://www.ti.com/lit/an/slla574/slla574.pdf?ts=1728906165655&ref_url=https%253A%252F%252Fwww.ti.com%252Flit%252FSLLA574

Or from Maxim with the MAX13487:
https://www.analog.com/media/en/technical-documentation/data-sheets/MAX13487E-MAX13488E.pdf

I will make a test setup in the next days and will post a message here if it works.

geronet1, 2024-10-14:
One question: Which issues do you get with the long cable? Only touch ring problems? Any communication issues?
If possible, get as much distance as possible between the signal cable and any 230V cables.

xxbyxx, 2024-10-15:
Hi, thanks a lot for your explanation. A documentation of your test setup would be highly appreciated! 
I looked for a datasheet/manual of R503 and found something here: https://download.mikroe.com/documents/datasheets/R503_datasheet.pdf

As I see it, the sensor uses semiduplex asychronism serial communication with a power supply of 3.3V - but please be aware that I am really not an expert in that field!