frickelzeugs/FingerprintDoorbell #97: Compile error with AsyncElegantOTA

ThomaxK, 2024-11-03:
I wanted to compile and upload the code via PlatformIO but got following error message:

Library Manager: Installing ayushsharma82/AsyncElegantOTA @ ^2.2.7
UnknownPackageError: Could not find the package with 'ayushsharma82/AsyncElegantOTA @ ^2.2.7' requirements for 
your system 'windows_amd64'

While looking up AsyncElegantOTA, i noticed that there is a newer version 2.2.9. But this didn't work neither.
I tried to update the code to ElegantOTA in async mode, beacause AsyncElegantOTA is deprecated but didn't get it to work.

Can someone help me? Thanks!

geronet1, 2024-11-03:
I've updated ElegantOTA in my fork:
https://github.com/frickelzeugs/FingerprintDoorbell/pull/98

ThomaxK, 2024-11-06:
Thank you, works great!

closed 2024-11-06 by ThomaxK