frickelzeugs/FingerprintDoorbell #104: "Your ESP8266 board is not supported." when using the web-flasher and wemos D1 mini module

Roellen, 2024-12-03:
Hi. 

I tried to use the web flasher app: https://frickelzeugs.github.io/FingerprintDoorbell-flasher/

When connecting to my Wemos D1 mini module it seems to connect (as the com port is shown), but then there is an error message:

**Your ESP8266 board is not supported.**

What can I do? Which ESP board will work?

tilly001, 2024-12-03:
Hello Roellen,

I use an ESP32 from azdelivery.

Search on Google:
[https://www.google.com/search?q=esp32+wroom+32+azdelivery&sca_esv=f29961f6c43f873e&sxsrf=ADLYWIL_cQUX2GQnnT01W2wEPY9jZ_3n5g%3A1733250750693&ei=vk5PZ_3vKbiS9u8PxrGwqAM&oq=esp32+wroom+32+azdel&gs_lp=Egxnd3Mtd2l6LXNlcnAiFGVzcDMyIHdyb29tIDMyIGF6ZGVsKgIIADIFEAAYgAQyBhAAGBYYHjIGEAAYFhgeMgYQABgWGB4yBhAAGBYYHjIGEAAYFhgeMgYQABgWGB4yBhAAGBYYHjIGEAAYFhgeMgYQABgWGB5I1RlQhwZY2hBwAXgBkAEAmAFgoAHPA6oBATa4AQPIAQD4AQGYAgegAo4EwgIKEAAYsAMY1gQYR8ICDRAAGIAEGLADGEMYigXCAgoQABiABBgUGIcCwgIKEAAYgAQYQxiKBcICCBAAGIAEGMsBmAMAiAYBkAYKkgcDNi4xoAf4LQ&sclient=gws-wiz-serp](url)

Roellen, 2024-12-03:
Ha! It works!
I found another ESP32 and this is working fine. 
(I thought it would run on wemos D1 mini, but it does not)

closed 2024-12-03 by Roellen