frickelzeugs/FingerprintDoorbell #110: Probleme nach 30 min Laufzeit...

Hafti210785, 2024-12-20:
Hi erstmal danke das ist genau das was ich gesucht habe 👍 ich hab alles soweit nach gebaut und war echt zufrieden. Hat top funktioniert. Auf meiner Werkbank. Habe dann alles in die Dose montiert und erstmal ging es. Aber nach ca 30 min wird die Berührung auf dem Ring nicht mehr erkannt. Teilweise braucht man zig Versuche oder muss vollflächig die Hand auflegen. Jetzt klingt es für mich nach einem Masseproblem. Aber es ist alles geerdet usw. Ich finde den Fehler nicht habt ihr eine Idee ?  Einmal stromlos machen warten neu starten und es geht wieder für ca 30 min. Es ist also kein generellen Problem.....  Ich habe alles hinter einem dcdc Konverter mit 3,3 Volt aber auch das sollte ja passen.

Noschvie, 2024-12-20:
Ist das Ding auch nach 30 Minuten ohne Unterbrechung per ping erreichbar?

Frosti1994, 2024-12-31:
Hey, habe das gleiche Problem, allerdings variiert die Zeit zwischen 5 - 45 min...

Hafti210785, 2024-12-31:
> Hey, habe das gleiche Problem, allerdings variiert die Zeit zwischen 5 - 45 min...

Ja variiert bei mir auch..   Hab mir jetzt einen neuen Taster bestellt muss mal kucken was passiert wenn ich den einbaue. An Erdung usw. Liegt es nicht hab alles durch probiert.

Noschvie, 2025-01-01:
> Hab mir jetzt einen neuen Taster bestellt

damit ist der R503 Sensor gemeint, oder?

Frosti1994, 2025-01-01:
> > Hey, habe das gleiche Problem, allerdings variiert die Zeit zwischen 5 - 45 min...
> 
> Ja variiert bei mir auch.. Hab mir jetzt einen neuen Taster bestellt muss mal kucken was passiert wenn ich den einbaue. An Erdung usw. Liegt es nicht hab alles durch probiert.

Ich könnte mir vorstellen, dass es ggf. mit der Firmware zu tun hat - muss da die Tage nochmal genauer einsteigen - hat jemand eine Idee, was der Maintenance-Zustand macht / sich verhält, bin ich gestern drüber gestolpert...

Frosti1994, 2025-01-01:
Habe gerade mal in den gelösten Issues gelesen und dort gab es das Problem, dass der Sensor nur funktionierte, wenn man Schuhe anhat - Lösung war, die Niederspannung per PELV abzusichern, also den Minuspol zu erden... Bin da kein Experte aber könnte die richtige Richtung sein!

Hafti210785, 2025-01-01:
Mh ja mal testen denn ein neuer Sensor hat nix gebracht hab's heute getestet

Hafti210785, 2025-01-02:
Finale 😜 ich habe es nun noch mal getestet indem ich wie von Frostie beschrieben den - der Spannungsversorgung geerdet habe. Scheint zu funktionieren es haben bei mir jetzt schon zig Personen probe geklingelt und ich habe keine Aussetzer.

closed 2025-01-02 by Hafti210785

Frosti1994, 2025-01-02:
Das freut mich sehr! Vielleicht kannst du mir  noch helfen... es hat bei mir nämlich nicht richtig geholfen! 

Habe den Minuspol geerdet, geklingelt, der Sensor hat mega schnell reagiert, aber das Klingeln hat nicht mehr aufgehört, also der Sensor hat durchgehend grün geblinkt, nachdem der Fingerabdruck erkannt wurde und in der Console wurde alle 2 Sekunden ausgegeben, dass der Fingerabdruck erkannt wurde, obwohl gar kein Finger auflag... Erst der Reset hat den Sensor wieder in den Normalzustand versetzt, nach erneutem Klingeln wieder das gleiche!
Polaritäten habe ich nicht vertauscht, also habe nicht ausversehen den Pluspol geerdet, aber das Gehäuse des Sensors habe ich geerdet - von dem Kabel habe ich der Einfachheit halber einen Abzweig direkt in der Klingel zum Minus auf meiner Platine gezogen...

Kann es daran liegen?

Frosti1994, 2025-01-02:
Erde vom Gehäuse getrennt - läuft seit 30 Minuten so stabil wie noch nie... ich freu mich! Danke!

Hafti210785, 2025-01-02:
Top 👍 --Gesendet mit der GMX Mail App

haselchen, 2025-01-24:
@Frosti1994 

Magst Du Deine Lösung mal bildlich zeigen?
Manchmal sagen Fotos mehr als Worte ;)

Frosti1994, 2025-01-25:
Hey,

wird nicht viel bringen, der relevante Teil ist im Verteilerschrank und ist eigentlich nur ein Kabel! ;)

Habe ein 5V-USB-Netzteil genommen und V- des USB-Kabels aufgeschnitten und per Steckverbindung eine Abzweigung zur Erde hergestellt.

Meine Testversion verzichtete auf Steckverbindungen.
Sprich, Kabel für Minus mittig abisoliert (nur abisolieren, nicht durchtrennen), Kabel als Y-Abzweig um den Draht gewickelt, bisschen verlötet, Schrumpfschlauch drüber, Abzweigung mit Erde verbunden, fertig!

Wichtig bei mir war es, die zusätzliche Erde direkt um den Sensor zu entfernen!

> [@Frosti1994](https://github.com/Frosti1994)
> 
> Magst Du Deine Lösung mal bildlich zeigen? Manchmal sagen Fotos mehr als Worte ;)