frickelzeugs/FingerprintDoorbell #112: Binary erstellen mit AsyncElegantOTA nicht möglich

actionbyte, 2024-12-28:
Nach einigen Versuchen scheitert es immer wieder an der korrekten Einbindung von dem AsyncElegantOTA. 
Da das Paket ohnehin abgekündigt ist, habe ich mit der GIT Anbindung versucht ElegantOTA zu verwenden, leider bekomme ich hier beim Compilieren einige andere Fehlermeldungen, welche ich nicht beheben kann.
Daher ist zumindest für mich dieser Abzweig vom FingerprintDoorbell leider nicht nutzbar.
Inwieweit man (@nobbi1991) den Code für ElegantOTA umschreiben könnte, bleibt für mich mangels Unkenntnis unklar, wäre aber für alle die eben nicht AsyncElegantOTA schon unter VSCode/PlatformIO eringebunden haben, eine große Hilfe. Glaube ich zumindest.

_Originally posted by @actionbyte in https://github.com/frickelzeugs/FingerprintDoorbell/discussions/107#discussioncomment-11668986_

actionbyte, 2024-12-28:
Beim Fork vom @nobbi1991 ist diese Änderung mit drin: 
[added option to configure mqtt port](https://github.com/nobbi1991/FingerprintDoorbell/commit/032244fd6e0efd8427ae86d7e7cb8759f1d1826b)

Im org fehlt mir die Möglichkeit den MQTT Port anzugeben weil ich auf dem Standartport einen anderen Adapter im IOB laufen habe und diesen eben ändern müsste.

geronet1, 2024-12-28:
Bei meinem Fork kannst du einfach den Port hinter den MQTT Servernamen schreiben:

192.168.0.12:1833

actionbyte, 2024-12-29:
Nice....aber ebenso ist AsyncElegantOTA nötig, welches ich nicht unter VSCode einbinden kann bzw. auch mit GIT Anbindung nicht automatisch beim Complilieren benutzt wird (da veraltet). Es wird eine Migration zu ElegantOTA empfohlen:

geronet1, 2024-12-29:
Nö, ist doch schon längst repariert:
https://github.com/frickelzeugs/FingerprintDoorbell/pull/98/commits/ad860eed3085ef14c965d8f0a1bb0512b43f2b58

Frosti1994, 2025-01-05:
@geronet1 Habe deine Fork geladen und wollte kompilieren und bekomme unendlich viele Fehlermeldungen - ein kleiner Auszug folgt (nur das Ende der Fehlermeldungen) - was mache ich falsch?

c:\.platformio\packages\toolchain-xtensa32\xtensa-esp32-elf\include\c++\5.2.0\ext\new_allocator.h:120:4:` error: invalid conversion from 'const char*' to 'unsigned char' [-fpermissive]
  { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
    ^
*** [.pio\build\esp32doit-devkit-v1\libe43\ESPAsyncWebServer\WebRequest.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\libe43\ESPAsyncWebServer\AsyncWebSocket.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\libe43\ESPAsyncWebServer\WebServer.cpp.o] Error 1
*** [.pio\build\esp32doit-devkit-v1\lib8f3\ElegantOTA\ElegantOTA.cpp.o] Error 1
============================================================ [FAILED] Took 7.60 seconds

geronet1, 2025-01-05:
Schick mal das ganze build-log über pastebin oder so

Frosti1994, 2025-01-05:
https://pastebin.com/NPDeqwVE

Die letzten Zeilen ändern sich bei jedem Buildversuch!

Frosti1994, 2025-01-05:
Ich versteh es einfach nicht - dachte erst, ein Problem mit dem PC, vor einiger Zeit hat mir Citrix alles zerschossen und ich konnte nichts kompilieren, solange es installiert war, aber auf zwei anderen Geräten die gleiche Meldung!

geronet1, 2025-01-05:
Schau mal Zeile 18-21, du hast ne neuere lib verwendet. Trag das ein dann gehts
https://github.com/geronet1/FingerprintDoorbell/blob/development/platformio.ini

Muss ich später mal schauen warum das nicht mehr geht

Frosti1994, 2025-01-05:
Habe genau die von dir verlinkte platform.ini benutzt! ;)
Trotzdem die ganzen Fehler... :(
Alle libs schon gelöscht, VSCode komplett runtergeschmissen und frisch installiert... und so weiter... ich weiß nicht, was es ist... habe nur deine Fork geladen, nichts verändert und auf Build geklickt... oder benötige ich noch besondere Extensions (abgesehen von platformio und C++)?

geronet1, 2025-01-05:
Probier mal mit der Version:
ayushsharma82/ElegantOTA@3.1.2

Frosti1994, 2025-01-05:
@geronet1 Dude, du bist genial, Danke <3

andy0172, 2025-01-08:
@geronet1  klasse Arbeit. Kannst du das auch auf den 8266 Portieren? Es gibt da schon einen Port aber mit altem OTA und ohne die ganzen Optionen. Wäre klasse. Danke Dir.

geronet1, 2025-01-08:
Hab mal alle libs geupdated, die platform war ja total alt..
Jetzt kompiliert es auch ohne Fehler: 
https://github.com/frickelzeugs/FingerprintDoorbell/commit/2dc7e3e14355d8b81a7e98ed00429d2c6e3f06ab

ESP8266 ist mir zu umständlich, kauf dir nen ESP32 :-)

andy0172, 2025-01-09:
Also seit dem neuen Update gibt es bei mir diesen Fehler beim Kompilieren:
src/main.cpp:10:22: fatal error: LittleFS.h: No such file or directory

******************************************************************
* Looking for LittleFS.h dependency? Check our library registry!
*
* CLI  > platformio lib search "header:LittleFS.h"
* Web  > https://registry.platformio.org/search?q=header:LittleFS.h
*

geronet1, 2025-01-09:
Gib mal über pastebin die Ausgabe, ein Full clean haste gemacht?

geronet1, 2025-01-09:
Achso Achtung! Da SPIFFS ja deprecated ist muss man evtl. den ESP komplett neu aufsetzen weil LittleFS dazu inkompatibel ist.

mahssel-dev, 2025-01-21:
gibts denn irgend jemanden, der irgendwo mal die neuen libs (gerne auch mit dem layout update von thedarkdeveloper) mal als komplettes repo gepusht hat?
Alles einzeln zusammen such (und am besten nix vergessen) ist dann doch etwas mühselig

geronet1, 2025-01-21:
Du meinst [TheMaskedDeveloper](https://github.com/TheMaskedDeveloper)? Die Teile sind bei mir alle drin

mahssel-dev, 2025-01-21:
das schon. dafür ist die ota lib deprecated.

was dann wieder dazu führt, dass ich mir eine lib lokal legen müsste und die "fehler" wegen deprecated wegarbeiten müsste ;)

geronet1, 2025-01-22:
Was neueres als ElegantOTA 3.1.6 sehe ich aber nicht? Erklär mal.

mahssel-dev, 2025-01-22:
Nein, da ich nicht davon ausgegangen bin, dass development = stable. Sondern, dass es sich bei Development um Inhalte handelt an denen du noch schraubst und die nicht zwangsweise korrekt laufen bzw. fertig sind. 

Wenn dev aber „fertig“ ist und läuft (auch mitm littlefs) dann vergiss meine Frage ;)

geronet1, 2025-01-22:
Ist bei mir seit einiger Zeit im Einsatz :-D

mahssel-dev, 2025-01-22:
alles klar.

dann muss ich nur noch schauen warum bei mit die LittleFS.h nicht vernünftig includiert ist ;)

das war der haupte grund glaub ich, warum ich das verworfen hatte.

halloworld007, 2025-02-22:
geronet wo finde ich deinen fork? 
bin ich blind?

Noschvie, 2025-02-22:
Ah... kein fork in eigentlichen Sinn, sondern offensichtlich eine Kopie / Clone plus Changes.